ut-robotics/picr22-team-valdis #3: [Mechanics] blue lines in sketches

Akustav, 2022-10-01:
Some sketches are left under defined. This can make it difficult to modify components later on and can create unpredictable situations when modifying things. This can be fixed by adding measurements to sketches until every line is black.

[image]

vircoh, 2022-10-02:
Võiks olla korras

Akustav, 2022-10-05:
Am I looking at the correct file? wall3 v15 has a lot of blue:
[image]

Ramp3 v4 has one hole underdefined:
[image]

closed 2022-11-05 by Akustav