ut-robotics/picr22-team-valdis #24: Consider using mountable XT-60 sockets

Akustav, 2022-11-26:
Maybe the use of a mountable XT-60 socket like this is good to power the robot.
[image]

[image]

vinku34, 2022-11-26:
do we have this kind of mountable connector or would I have to make a holder to make it mountable?

Akustav, 2022-12-04:
They are available in the lab

vinku34, 2022-12-05:
I made a variant with these mountable sockets. The current issue would be on the side with the NUC power cable. Right now I can move it closer to the edge a bit but im not 100% sure if it will clear the power cable or not.
[image]

Also the holes in the rear acted as a place to shove the battery voltage displays. I have an Idea to mount them here with 3D printed boxes and screw in from the inside with the same flush fit as most of the bolts I have. with little indents for nuts so that I dont have to thread plastic
[image]

Akustav, 2022-12-07:
Seems reasonable. Also FYI, the XT-60 sockets themselves have threaded inserts, so you just need a 3mm hole to mount them. Your current holes are 2.5mm

closed 2022-12-14 by maidlam