ut-robotics/picr22-team-valdis #38: Mounting hole too close to components

ReikoR, 2022-12-02:
I added a cylinder with a diameter of 5.5 mm (same as M3 bolt head) on the PCB.

[image]

ReikoR, 2022-12-04:
Other mounting hole might be too close to the motor output pads or might damage the solder mask and short the copper below.

[image]

vinku34, 2022-12-04:
Can I just keep the one mounting hole then? It won't be supporting much weight anyways.

ReikoR, 2022-12-04:
It would be better to have at least 2 mounting holes to prevent the PCB from rotating.

There is room near the original location, it was just too close to the edge before. Silkscreen text needs to be moved. Silkscreen text can also be smaller if there is not enough room.

[image]

closed 2022-12-04 by ReikoR